apache/plc4x #2101: [Feature Request]: Max Modbus Registers per Query

parthgoel06, 2025-05-21:
### What would you like to happen?

It would be great if there were a setting to limit max Modbus registers from 125 (default) to something like 40 or 50. Based on this, the ModbusOptimiser can break the queries into multiple groups. This is needed because some devices have register limiters.

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] S7

IsmoLeszczynski, 2025-05-21:
I believe this is available as a connection string parameter in 0.13.0-SNAPSHOT, have you tried it?

https://plc4x.apache.org/plc4x/pre-release/users/protocols/modbus.html

`max-registers-per-request`

parthgoel06, 2025-05-21:
There's a separate pre-release documentation too😲. My bad, I did not know about it. I'll give this a try and close this request if it works.

closed 2025-05-21 by parthgoel06