apache/plc4x #2102: [Feature Request]: Force Function Code 6 when writing modbus tag

parthgoel06, 2025-05-21:
### What would you like to happen?

Is there a way that I can force the use of ModbusPDUWriteSingleRegisterRequest (Function Code 6) instead of ModbusPDUWriteMultipleHoldingRegistersRequest (Function Code 16) when I write a ModbusTCP tag? My device only accepts function code 6 when writing.

### Programming Languages

- [x] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2025-05-22:
HI,

should be quite easy to implement, if you want to give it a shot, I'm happy to help you scratch your own itch. 

If not, there should also be some people in the community willing to help you out with this. 

If you want me to do it. I stopped fixing and implementing stuff for free, so feel free to leave a donation here: https://buymeacoffee.com/christoferu (Doesn't have to be a huge one), then I'll take care of it.

Chris

parthgoel06, 2025-05-22:
I would like to give implementation a shot, but would this implementation be done in the PLC4X library, or can I handle something in my service itself?

I have figured as much ->
In `plc4j/drivers/modbus/src/main/java/org/apache/plc4x/java/modbus/base/protocol/ModbusProtocolLogic.java` in `getWriteRequestPdu` function, when the tag is an instance of `ModbusTagHoldingRegister`, it defaults to `ModbusPDUWriteMultipleHoldingRegistersRequest`, which is a write request of function code 16.

chrisdutz, 2025-05-22:
Well one option would be to do it on your application and simply create one item per request. However am I not sure that I didn't "clean up" the driver code to simply use the multi element option.

The more sustainable way (that others having the same problem would profit from) would be to add a cinfig option to the plc4x modbus driver implementation and update the protocol logic.

parthgoel06, 2025-05-23:
So I am sending only 1 16-bit register per write request, but it's still writing with function code 16 (multiple register write). It works for most cases, but some manufacturers do not allow function code 16 for single register write.

Yeah, it would be great to have a config option in the address at the time of writing, where we can choose single or multi write.

I am currently going through the library code, but I'll definitely consider donating also 😁.

chrisdutz, 2025-05-23:
Yeah ... I mentioned, that I might have "simplified" the driver to only use the one type as I couldn't see a benefit of having both (According to the Spec both should be supported, but it seems especially with Modbus, everyone does as he pleases).

So I guess you'll have to update the driver, or have someone update it for you.

parthgoel06, 2025-05-23:
@chrisdutz I just donated you a coffee. If possible, can you do this implementation? 😁

chrisdutz, 2025-05-23:
So ... I would assume that I need to spend 1-2 hours on this fix and you just donated 5$ for that ... admittedly I'm slightly speachless.

parthgoel06, 2025-05-23:
I would have donated more, but I already donated a good amount on Open Collective to PLC4X contribution and started the donation, just because I wanted to support you guys.

chrisdutz, 2025-05-23:
Well ... just checked and yeah ... thanks for the open-collective donation ... didn't see that connection, ... I'll have a look if I can squeeze it in somehow. I do know that finances are not spread out fairly over the globe. It just felt as if me fixing things wasn't worth much.

parthgoel06, 2025-05-23:
Thanks and sorry if you felt that way, it was not my intention. Also, as I mentioned before to you, I'll try to get the company that I work for to donate a larger amount if they agree to it.

chrisdutz, 2025-05-24:
So yesterday I spent about an hour on the issue.

It's easy to implement something that uses the only one register method, if the tag is only one register big anyway. However, if it should break down multi register tags, that weighs be tricky and take a lot longer to implement.

Would that be a compromise?

parthgoel06, 2025-05-24:
Also, I am not in much hurry, so you can take as much time as you want. I am also willing to help with this implementation if you can guide me. I tried to go through the library code but could not link how the higher-level APIs link to the protocol code.

chrisdutz, 2025-05-24:
Well ... the option I proposed, would be simple to implement as I don't have to translate tags into intermediate tags. If I want to support this feature for all types of requests, I have to start doing request rewriting ... that's not something I do in 1-2 hours ... I think it's more something like the new Modbus Optimizer, that I built and that would be something down the lande of a full day to implement. 

So it's not only about the time till it's finished ... it's about financing the work to do it. 

I am currently implementing my proposed option as this is a quick win, that would allow you to communicate with your device, but forgive me that this is already 20 times more work than I would be willing to invest for 5$.

chrisdutz, 2025-05-24:
Your bigger proposed option would require the user to actively select an option in the connection string. In general, the work I'm currently doing is a prerequisite for this to work. So in this case the user selects the "single-register-writes" option and then the opimizer will let single-register tag writes pass without modification, but if you write a tag that is larger than one register, it will behind the scenes rewrite the tag into a series of single-register tags execute them and then merge the results back together. This should work on all devices ... but it's a lot more work to implement. And I wouldn't be willing to do that part for less than 400€ (I seem to have to add real price tags in order to avoid situations like this)

parthgoel06, 2025-05-24:
Well, thanks for your work, it's really appreciated and the solution that you have implemented currently works too. And surely I'll keep in mind the donation part for the future.

chrisdutz, 2025-05-24:
So ... my changes are committed and if you pull them you should be able to use the single register reads.

[image]

parthgoel06, 2025-05-24:
It's working now, thanks a lot again. I'm closing this thread now.

closed 2025-05-24 by parthgoel06