apache/plc4x #2229: [Bug]: Modbus - Number of threads repetitively increase at each reconnection.

frankpuppa, 2025-08-27:
### What happened?

While I was testing the stability of a modbus application based on plc4x, I faced the problem that on successful reconnection attempts the number of live threads keeps increasing continuously. Specifically, there is a pool as such: pool-N-thread-M (probably related to netty) which sits waiting and does not shut down. 
The same applies when using the CachedPlcConnectionManager
Connection string: modbus-tcp://127.0.0.1:9081?unit-identifier=1

Here is a sample code with springboot:
```
 @Scheduled(fixedDelayString = "${scheduling.interval}") 
    public void update() {
        logger.info("Executing update method...")

        try {
            reconnect(false);

            if (plcConnection == null || !plcConnection.isConnected()) {
                logger.info("PLC connection is not alive, exiting...");
                return;
            }

            logger.info("OK");
        } catch (Exception e) {
            logger.error("Error while trying to read ... reconnecting", e);
        }
    }

      protected void reconnect(boolean error) {

        try {
            try {
                if (plcConnection != null && !plcConnection.isConnected()) {
                    logger.info("Reconnecting...");
                    // either with plcConnection.connect() or
                    // connectionManager.getConnection(plcConnectionString); There is a n increment
                    // of threads
                    // plcConnection = connectionManager.getConnection(plcConnectionString);
                    plcConnection.connect();
                    return;
                }
            } catch (Exception e) {
                logger.error("Error: {}", e.getMessage());
                return;
            }

            // Open a new one
            if (plcConnection == null) {
                logger.info("Connecting...");
                plcConnection = connectionManager.getConnection(plcConnectionString);
            }
        } catch (Exception ex) {
            logger.error("Error while trying to reconnect to the plc");
        }
    }
```
The function runs with a 10 seconds interval, in this time window I simulate the disconnection/reconnection of the cable.
I'm attaching a small video which shows this behavior more clearly. 

I understand that this is a corner case but it can lead to a waste of resources which I would like to avoid.

https://github.com/user-attachments/assets/003b4d80-e600-498a-b7bd-44c8697c4bd9

Thank you.

### Version

v0.13.0

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [x] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [ ] S7
- [ ] S7-light

frankpuppa, 2025-08-27:
Hello @glcj,
calling close on a disconnected connection raises an exception. Which eventually must be reopened using the 
```
plcConnection = connectionManager.getConnection(plcConnectionString);
```
I had tried this and under these circumstances, when the new connection opens correctly a new pool is created while the previous sits waiting.. 
I also tried to manually close the DefaultNettyPlcConnection as such:
```
EventLoopGroup group = ((DefaultNettyPlcConnection) connection).getChannel().eventLoop().parent();
group.shutdownGracefully();
```
but it did not help..

chrisdutz, 2025-08-27:
So let me understand your test-case a bit more ... you're connecting to a simulator and polling that in regular intervals. As soon as you turn it off (or in a real device, pull the plug or turn it off) things start failing (obviously) but as soon as you make the device available again, it doesn't go back to normal?

chrisdutz, 2025-08-27:
But admittedly with respect to maintainability, Netty is a nightmare. That's why I'm already working on a completely Netty-free core for PLC4J ... I do hope the funding, that I applied for, gets granted. Then all of this work will go into PLC4J. If not ... well ... it's go into my commercial offering (Which I hope to have available within a few months)

frankpuppa, 2025-08-27:
> So let me understand your test-case a bit more ... you're connecting to a simulator and polling that in regular intervals. As soon as you turn it off (or in a real device, pull the plug or turn it [...]

Hello @chrisdutz 
This is exactly the test case. I need to make sure that when the device is available again,  everything goes back to normal. At the moment, it does somehow, apart for the number of threads that keep increasing 😄. 
While I was investigating the issue I had imagine that it was related to Netty as it seems that it is a common issue with this library (at least from a quick search on the web). Anyway, thank you for letting me know of a possible netty free solution.

chrisdutz, 2025-09-03:
Let's see if Franks PR will solve this problem?

frankpuppa, 2025-09-03:
Hello, 
Actually I haven't used the CachedPlcConnectionManager and I did not try to call removeCachedConnection. What I did noticed while using the CachedPlcConnectionManager, is that once the channel becomes inactive, the connection retrieved by the pool was always broken even if the device was put back online (no new attempt was made). Perhaps calling removeCachedConnection, may solve the problem of getting a fresh connection but under these circumstances resources may still leak.. Definitely something worth trying..

chrisdutz, 2025-09-30:
Hmmm ... I thought we updated things to close the executor service of discarded connections.

Could you please whip up a PR with your changes so we can have a look? But admittedly I'm always super cautious with singletons. They usually are always the simple way out but always come with huge costs in maintenance. (Like running multiple test-suites in parallel from the same VM (As is usually happens in Maven) ... or if you're running many, many connections on one machine. I would be super careful with them. I think they even break things when using OSGI (I know there are some users/integration modules that rely on the "kar" files we produce)

chrisdutz, 2025-09-30:
Possibly an option, where the DriverManager passes in such an executor would be a preferable option.

For me the only long-term sustainable option is to rip out Netty and rebuild PLC4X without it. But considering the amount of volunteers willing to work on this, I don't think that's ever gonna happen.

chrisdutz, 2025-09-30:
Yeah ... so unfortunately as it relies on statics/singletons and I know these are bad for OSGI based options, I would be hesitant to merge that PR. @splatch might have more info. I think he mentioned it results in multiple classloaders beind required.  

If you find a way to create it in the Driver manager or the Driver implementation and pass it down the stack, that would be the better option.