apache/plc4x #2240: [Bug]: Fix the broken Docker tests for OPC-UA

chrisdutz, 2025-09-02:
### What happened?

I have disabled the Docker based OPC-UA tests for now. It seems after updating to Java 21 there were some issues. I have fixed them locally, however it seems Jenkins doesn't pick them up, so I'm disabling the tests until someone has the cycles to fix them.

### Version

0.14.0-SNAPSHOT

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [ ] Modbus
- [x] OPC-UA
- [ ] Profinet
- [ ] S7
- [ ] S7-light

klaus321901, 2025-10-03:
Hi, I’m a newcomer and would like to take this issue. I’ll look into fixing the Docker-based OPC-UA tests so they work again with Java 21.

chrisdutz, 2025-10-03:
Hi Klaus, 

I see you seem to be very interested in this project and participating in various areas. 

Would you be so kind and sign up on our mailing list dev@plc4x.apache.org (You sign up by sending an empty email to dev-subscribe@plc4x.apache.org and replying to the email you'll get after that).
We'd love to get to know you. Usually we mentor new people and would like to get to know you.

Chris