apache/plc4x #2246: [Bug]: S7 Read Request gets timed out after approximately 5-10 minutes

shekhartanwar, 2025-09-05:
### What happened?

**Issue**
After upgrading to library version 0.13.0, my application, which reads 207 memory blocks from a PLC using the S7 protocol every second, starts throwing timeout exceptions after running for 5–10 minutes. With version 0.12.0, the application runs without issues.

**Expected behavior**
The application should continuously read all 207 memory blocks every second without timeout exceptions, as it does with version 0.12.0.

**Steps to reproduce**
1. Use library version 0.13.0.
2. Continuously read 207 memory blocks from a PLC every second.
3. Observe that after 5–10 minutes, timeout exceptions occur.

**Additional context**
In 0.12.0, I manually divided the 207 memory blocks into ~21 PlcReadRequests (max 20 blocks per request) using batching.
In 0.13.0, batching is handled internally by the library.
No other changes were made to the application code.
This issue blocks me from upgrading to 0.13.0.

**Environment**
Library version: 0.13.0 (works in 0.12.0)
Protocol: S7
Number of memory blocks: 207
Frequency: 1 read per second

### Version

v0.13.0

### Programming Languages

- [ ] plc4c
- [ ] plc4go
- [x] plc4j
- [ ] plc4net
- [ ] plc4py

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] C-Bus
- [ ] CANopen
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] IEC-69870
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] Profinet
- [x] S7
- [ ] S7-light

sruehl, 2025-09-05:
try 0.13.1

chrisdutz, 2025-09-05:
Well ... does it generally work, or are you just seeing that in the logs and generally everything is working?
In general S7 devices are some times a little "bitchy" ... if they don't like something, they just hang up. So some times a bit might flip or the S7 thinks one flipped and it just hangs up .... especially under heavy network loads, I've seen that happen.

Are you using the connection cache or are you opening a connection for every request? I hope you're using the connection-cache, as that should handle the reconnects and also improve performance quite a bit.

shekhartanwar, 2025-09-05:
> Well ... does it generally work, or are you just seeing that in the logs and generally everything is working? In general S7 devices are some times a little "bitchy" ... if they don't like something, [...]

I am creating only 1 instance of PlcConnection and using it in all read/write requests, i have created a getConnection() method that whenever called checks if plcConnection.isConnected() then accordingly reconnects if not connected. But at the end, the connection object is Singleton. This connection stays open till it's automatically disconnected.
As fas as i understood, for using Connection Cache, i have to close connection every time so that all other threads can use it.

chrisdutz, 2025-09-05:
Hi,

yeah ... well you "technically" close it after doing your thing, but it doesn't really close, it but return it to the cache (and potentially waiting threads).
In your case I guess you have implemented part of the functionality. I just wanted to make sure that you don't expect the connection to be open indeffinitely. If you're checking things and potentially re-creating connections, you should be fine. But I guess your log output is simply the log of an operation failing and your code then creates a new connection the next time.

shekhartanwar, 2025-09-09:
> Hello, Can you tell me how many bytes (on average) you're reading from each DB, and if it's not a problem, the CPU model with the assigned communications load? I'm in a remote location without much connectivity, but I have access to a CPU315 2PN/DP. I'm interested in evaluating that test. Best regards,

I am not sure of PLC Configuration because that is managed by some other team. I read approx. 250 bytes per second from single memory block with different offsets. **Correction:** Single DB with different offsets.