apache/plc4x #2247: does plc4j-driver-s7 0.13.0 not support s7-200

john1337, 2025-09-06:
### What would you like to happen?

i am using plc4j-driver-s7 0.13.0 to communicate with simens s7-200,but it can't be connected,but when i am using s7connector,everything works well,there's nothing wrong with network
example codes:
            // 1. 创建连接
            PlcConnection connection = PlcDriverManager.getDefault()
                    .getConnectionManager()
                    .getConnection("s7://172.16.8.249:102?transport-protocol=pniop&remote-rack=0&remote-slot=2"
                            + "&controller-type=S7_200");

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2026-08-16:
Actually it seems that the current driver is able to talk to S7-200 https://github.com/apache/plc4x/issues/2641

closed 2026-08-16 by chrisdutz

chrisdutz, 2026-08-16:
This is the same ground as 2641, and the S7
page now documents it (`s7.adoc:335`, "Siemens S7-200 / S7-200 Smart"), which states that reading
and writing were confirmed against an S7-200 Smart. The reporter's string
`s7://172.16.8.249:102?transport-protocol=pniop&remote-rack=0&remote-slot=2&controller-type=S7_200`
is broken in three ways, any one of which is enough to stop it working:

1. `transport-protocol=pniop` - `S7Driver.getSupportedTransportCodes()` returns only `cotp`.
2. `remote-rack` / `remote-slot` are transport parameters and need the transport prefix
   (`cotp.remote-rack`). `ConfigurationFactory.createPrefixedConfiguration` only passes through
   parameters starting with `cotp.`, so unprefixed ones are silently ignored and the defaults stay.
   Their `remote-slot=2` never reached the transport at all.
3. The TSAPs are missing. Per the documentation an S7-200 (Smart) needs
   `cotp.local-tsap=512&cotp.remote-tsap=512`, which is the part that actually makes the connection
   come up.

`controller-type=S7_200` was the one part they had right. The working form is
`s7://172.16.8.249?cotp.local-tsap=512&cotp.remote-tsap=512&controller-type=S7_200`. Note that
`S7_200` deliberately reports no UserData services (`supportsUserDataServices`), so browse, alarms
and cyclic subscriptions are not available on it - reads and writes are.